CRESYM/colib0.github.io #169: [COLIB-NGC] T1.1 update GovCT2

martinscripts, 2024-06-20:
### Do you want to modify a model or test case?

- [X] Model
- [ ] Test Case

### What is the impact of modification to the model or test case?

- [X] minor
- [ ] moderate
- [ ] major

### Briefly describe the modification to the model/ test case

typo, folder structure

martinscripts, 2024-06-20:
The main change is 
- a fix in the GovCT2 folder structure
- a suggested update to the folder structure for governors, avr and pss

please comment on whether you agree to this and/or have other suggestions. 

- please also note that i change d the names of the other Governor files, I think this improves readability in the overview. 
- if you agree, I can add the same alteration to the PSS and AVR names

GJCRESYM, 2024-06-27:
Hi Marin, 

Could you please create the pull request to "dev" branch instead of main?

martinscripts, 2024-06-27:
Hi @GJCRESYM, I changed base of the existing PRs, it should work now. BR

GJCRESYM, 2024-11-22:
@ErwanGuichard49 , @martinscripts : The changes have been merged and are live on Colib.net. Please verify and let us know if any modifications are required. Thank you.

BR,
Geethu

GJCRESYM, 2024-11-22:
Changes are merged to main

closed 2024-11-22 by GJCRESYM