CRESYM/colib0.github.io #200: [Model/ Test case Name]: Induction motors

FredericSabot, 2024-08-09:
### Do you want to add a new model or test case?

- [X] New Model
- [ ] Test Cases

### Briefly describe about the contribution

This issues adds a simple single-cage induction motor model and the WECC fifth order motor model.

GJCRESYM, 2024-08-12:
Hello @FredericSabot, 

Thank you for contributing to Colib. If you need to know the standard procedure, please follow https://colib.net/pages/about/contribute.html and https://colib.net/pages/about/authorGuidelines.html.

You can refer to the model template: https://colib.net/pages/templates/modelTemplate.html

Please plan to create a pull request to **dev** branch once you are ready with your model. 

Thanks & regards,
Geethu Joseph
geethu.joseph@cresym.eu

FredericSabot, 2024-08-12:
Hello Geethu,

I have created the PR https://github.com/CRESYM/colib0.github.io/pull/205.

Best regards,
Frédéric

GJCRESYM, 2024-08-12:
Thank you for you PR @FredericSabot.

Can you downgrade the ruby version to 2.6.5 as mentioned in the readme file: https://github.com/CRESYM/colib0.github.io#

If you have followed the steps in Readme, it should avoid the issues that you have now.

BR, 
Geethu

FredericSabot, 2024-08-12:
Installing ruby-2.6.5 on Ubuntu is a bit tedious: https://github.com/rvm/rvm/issues/5263 but it works.

I updated the PR and it knows displays as expected.

PR #215 (2024-11-15): Add induction motor models
Closes #200
closed 2024-11-15 by GJCRESYM