CRESYM/colib0.github.io #221: Point-to-Point HVDC EMT VSC

Lampros-Cresym, 2024-09-11:
### Specify the model/test case you evaluated

- [X] Model
- [ ] Testcase

### Name the model/test case ID, you would like to evaluate

0000

### What is the impact to the model or testcase?

- [ ] minor
- [X] moderate
- [ ] major

### Briefly describe your opinion about the model/testcase

- Section 2 typo: "time step cannot be **increased** too much"
- Is the VSC model an MMC one or a 2-level VSC? It would be good to add a sentence on this.
- It is not clear for me if this is an EMT model, an RMS model or an RMS model that could potentially be used also in EMT simulations. The title says "EMT" but the implementation is in dynawo, which is mostly RMS.
- Figure 1 shows a symmetric monopole topology (and not a bipole one).
- Please check the equations for the DC voltages and align with the current convention in Figure 1. It seems to me that it should be DV_dc1 = (Idc1-Idc)/Cdc and DV_dc2 = (Idc-Idc2)/Cdc
- Does the capacitor include the converter capacitance (in case of 2-level VSC) or just the line one?
- The implementation is unclear. Is the grid forming implemented or both VSCs are frid following?
- To be discussed, but maybe this fits better as a test case. I would keep here only the description of the model of the VSC and add a link to the test case.

GJCRESYM, 2024-11-28:
@Carlos-Alegre : Do you have any updates on this issue?

GJCRESYM, 2024-12-02:
@Lampros-Cresym : Could you please have a look?

Lampros-Cresym, 2024-12-03:
Sorry, but it's impossible for me to understand what could be the problem just by the figure. Is the final steady state in the simulation you sent me the same as the one you tried to impose in the beginning? Is this transient (which seems quite big) only that much in the reactive power? Could you perhaps try to isolate the different subsystems to try to locate which component is responsible?