CRESYM/colib0.github.io #228: [Bug]: References are not displayed in Author Guidelines page

GJCRESYM, 2024-09-19:
### Is there an existing issue for this?

- [X] There are no existing issues

### Priority of this issue

- [ ] high
- [X] medium
- [ ] low

### Describe the current behaviour

Current behaviour:
[image]

References are not getting displayed

### Describe the expected behaviour

Display the expected result!

### Steps to Reproduce

1. Go to
2. Click on
3. Enter ...
# Add more steps here,

Buqra55, 2024-09-19:
try this
https://bit.ly/4gwC82r
I put the necessary dlls in the archive

GJCRESYM, 2024-12-19:
The issue is fixed and merged to main

closed 2024-12-19 by GJCRESYM