CRESYM/colib0.github.io #232: Generic EMT Grid Following Voltage Source Converter with SVPWM Modulation

Lampros-Cresym, 2024-09-20:
### Specify the model/test case you evaluated

- [X] Model
- [ ] Testcase

### Name the model/test case ID, you would like to evaluate

163

### What is the impact to the model or testcase?

- [X] minor
- [ ] moderate
- [ ] major

### Briefly describe your opinion about the model/testcase

- Problem with display of inline equations in Sub-section 3.5
- We would need to have to more explanation of the current limitations
- It is not clear for me what is the advantage of modeling the SPWM in this model. I understand that it has a representation of the modulation, but (as noted) the rest of the model does not make it suitable either for harmonics or for unbalanced situations. I am not proposing to remove the SPWM, but I would like to have also the version of having the model without the SPWM (to also make a clear correspondance with the EMT grid forming model)

Carlos-Alegre, 2024-10-09:
- Fixed the equation issue.
- Noted for this and the rest of the models.
- The case without modulation is basically the same but just using directly the output of the Current Loop directly as the generated voltage of the VSC. It is explained in the last paragraph of section 3.3, although it can be further explained if it is unclear. Other than that, I would say the modulation can be interesting even though no unbalance or no harmonics coming from the PLL are modelled. 
- The implementation of the model itself are open-source, even though it is true the language is closed. 

We can comment these points in the meeting, as some of them can be applied to other models

Lampros-Cresym, 2024-12-05:
Issue closed, model has been replaced with "Generic EMT Grid Following Voltage Source Converter"