CRESYM/colib0.github.io #234: GENTPF/GENTPJ Synchronous Machine with Saturation

Lampros-Cresym, 2024-09-23:
### Specify the model/test case you evaluated

- [X] Model
- [ ] Testcase

### Name the model/test case ID, you would like to evaluate

196

### What is the impact to the model or testcase?

- [X] minor
- [ ] moderate
- [ ] major

### Briefly describe your opinion about the model/testcase

- Typo first paragraph of section 2: "later" --> "latter"
- Sub-section 3.1 - Parameters: It would be very nice to have a range of values for each parameter in the table
- Missing the definition of the $\omega_m$ and $Sat_q$ variables. In fact, I think that $\omega_m$ should be replaced simply by $\omega$ in all equations
- I think there is a mixup between the $Sat_d$ and $Sat_q$ variables in the $$dE''_q/dt$$ and $$dE''_d/dt$$ equations
- I think $X''_q$ should be replaced by $X''_d$ here
[image]

Carlos-Alegre, 2024-10-09:
Added suggested corrections. Pending adding ranges for the parameter table

GJCRESYM, 2024-11-28:
@Carlos-Alegre : Are you working on this issue now? Do you have any branch created. If yes, could you please the branch to this issue. 

Thanks in advance

Carlos-Alegre, 2024-11-28:
Done! I left it out when doing the PRs

GJCRESYM, 2024-12-19:
Merged

closed 2024-12-19 by GJCRESYM